Wiz-IO/framework-quectel #1: BC66 I2C (BME280)

Redferne, 2019-10-30:
Using the Olimex Dev Kit (https://www.olimex.com/Products/IoT/NB-IoT/NB-IoT-DevKit/open-source-hardware). Attached a BME280 to the I2C0-SCL(RI) and I2C0-SDA(DCD) pins but can only read zeros from the I2C bus. Tried both address 0x76/0x77. Took a quick look with the logic analyzer and it seems that it's driving the bus at 400kHz (as expected). But it's decoding a NACK?

Using firmware BC66NBR01A04, is it worth testing BC66NBR01A07?
Any other ideas? Thanks!

[image]

Wiz-IO, 2019-10-30:
Arduino ? OpenCPU?

Redferne, 2019-10-31:
Yes! Works beatifully, thanks. I would argue that the Arduino Wire interface should be using 7bit (https://www.arduino.cc/en/reference/wire). Will you accept PR?

Wiz-IO, 2019-10-31:
8 bits is from Mediatek system driver, I just follow Quectel SDK ...
what is PR?

Wiz-IO, 2019-10-31:
Pool Request ? ... no problem...
if someone find some isues ...

Redferne, 2019-10-31:
Pull Request (github patch). Will spend some hours on this devboard (Arduino framework) and see if I can make something cool. Will submit PRs. Thanks for your huge work on this! It is really amazing.

Redferne, 2019-11-07:
PR merged :smile:

closed 2019-11-07 by Redferne