Wiz-IO/framework-quectel #2: BC66 Arduino Platformio micros()

Redferne, 2019-10-31:
Saw this: https://github.com/Wiz-IO/Arduino-Quectel-M66/issues/1

Using micros();

```
PLATFORM: Quectel 2.0.14 #ba06662 > Olimex NB-IoT-DevKit
HARDWARE: ARM, MT2625 78MHz, 97.66KB RAM, 195.31KB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES: framework-quectel 02002ee, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 12 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <GxEPD> 3.0.8 #10187e3
|   |-- <Adafruit GFX Library> 1.6.1 #fc13f34
|   |   |-- <SPI> 1.0
|   |-- <SPI> 1.0
|-- <Adafruit GFX Library> 1.6.1 #fc13f34
|   |-- <SPI> 1.0
Linking .pio/build/bc66/program.elf
/home/x/.platformio/packages/framework-quectel/libraries/lib_BC66R01A04V01.a(api.o): In function `api_get_micro':
api.c:(.text.api_get_micro+0x6): undefined reference to `hal_gpt_get_free_run_count'
collect2: error: ld returned 1 exit status
*** [.pio/build/bc66/program.elf] Error 1
```

Wiz-IO, 2019-10-31:
hal_gpt_get_free_run_count()
is removed from API ... I don't remember why

for version 4
//    .globl __hal_gpt_get_free_run_count_veneer
//    .thumb_func
//__hal_gpt_get_free_run_count_veneer: ldr pc, =**0x081012E1**

Wiz-IO, 2019-10-31:
will test this days - if work good will update libs

Redferne, 2019-10-31:
Great. Have you tested SPI on BC66? Trying to drive a e-paper display, and I'm seeing some issues with timing and byte transfers.

Wiz-IO, 2019-11-01:
SPI is not tested
https://github.com/Wiz-IO/framework-quectel/blob/master/opencpu/bc66/include/ql_spi.h

closed 2020-04-08 by Wiz-IO