Wiz-IO/framework-quectel #3: Psm.wakeup(x)

Redferne, 2019-11-07:
BC66 Arduino PlatformIO.

Using this in main.cpp
```
  Psm.wakeup(x);
```

Yields:
```
/home/x/.platformio/packages/framework-quectel/libraries/lib_BC66R01A04V01.a(api_rtc.o): In function `api_rtc_start':
api_rtc.c:(.text.api_rtc_start+0x50): undefined reference to `user_rtc_callback'
```

Wiz-IO, 2019-11-07:
void user_rtc_callback(void *user_data) 
{
      //userware
}

Wiz-IO, 2019-11-07:
Quectel had a bugs a year ago (ver 4)
I don't know if they fixed it....
The problem was that after 10 PSM wakeups the module crash

or try default API
https://github.com/Wiz-IO/framework-quectel/blob/master/opencpu/bc66/include/ql_rtc.h
https://github.com/Wiz-IO/platformio-quectel-examples/blob/master/OpenCPU/bc66/example_rtc.c

Wiz-IO, 2019-11-07:
BTW: 
The PSM is the "hard" part...
and Quectel don't know how it works :) ... ask support

Redferne, 2019-11-07:
> BTW:
> The PSM is the "hard" part...
> and Quectel don't know how it works :) ... ask support

Yeah. PSM is tricky, you need network support also. I got something semi stable using Psm.set(300, 10, 1) It does however not wakeup exactly after 300s, it is very random, between 300 and 3600s But it wakes up :smile: been running on battery over the weekend and it is still running :laughing:

Wiz-IO, 2019-11-07:
to enter in PSM module must go in IDLE task more times
also depends on the network, ours is a constant times and it is difficult for me to try variants

Redferne, 2019-11-13:
Upgraded to R01A07 but still having issue with PSM and RTC periodic wakeup.
main.cpp
```
void user_rtc_callback(void *user_data);
void user_rtc_callback(void *user_data)
{
    DEBUG_PSM("\n[PSM RTC]\n");
}
...
  Psm.wakeup(300);
  Psm.Sleep();
```
Compile output:
```
/home/x/.platformio/packages/framework-quectel/libraries/lib_BC66R01A07.a(api_rtc.o): In function `api_rtc_start':
api_rtc.c:(.text.api_rtc_start+0x2e): undefined reference to `rtc_sw_timer_start'
api_rtc.c:(.text.api_rtc_start+0x50): undefined reference to `user_rtc_callback'
```

Wiz-IO, 2019-11-13:
Hi,
I not have time (not) to test
Try use original API RTC ( must work )
https://github.com/Wiz-IO/framework-quectel/blob/master/opencpu/bc66/include/ql_rtc.h#L144

Wiz-IO, 2019-11-13:
https://github.com/Wiz-IO/platformio-quectel-examples/blob/master/OpenCPU/bc66/example_rtc.c#L117

Wiz-IO, 2019-11-13:
try this:
```
// Psm.wakeup(uint32_t seconds, sleep_callback_t userCB)
// if userCB = NULL
extern "C" /* from CPP */ void user_rtc_callback(void *user_data)
{
  // empty
}

// if userCB is not NULL
void psm_sleep_callback(void *a, uint32_t b)
{
    // empty
}
```

Psm.wakeup(300, **psm_sleep_callback**);

Wiz-IO, 2019-11-13:
edit:
Psm.wakeup will register callback beforeSleep
if is NULL - there exist default,  if is not NULL you need make any as default
https://github.com/Wiz-IO/framework-quectel/blob/master/arduino/cores/bc66/PSM.cpp#L124

just need only:
```
extern "C" /* from CPP */ void user_rtc_callback(void *user_data)
{
  // empty, .... is afterWakeUp, not need, module will "restart"
}
```

Redferne, 2019-11-13:
Ahh! Thanks for the idea, I got it working using RTC:
```
  RTC.timer_begin(rtc_callback);
  RTC.timer_start(300000, true); // Wake up from PSM every 5 min.
  Psm.Sleep();
```
The main problem was that EDRX was enabled, so I disabled it with AT+CEDRXS=0.
Now PSM behaves a little better :smile: 

Will push a PR, when I clean up the code.

closed 2020-04-08 by Wiz-IO