Wiz-IO/framework-quectel #9: could not initialize project

manusoman92, 2020-07-03:
PIO Core Call Error: "The current working directory /home/manu/Documents/PlatformIO/Projects/sample will be used for the project.\n\nThe next files/directories have been created in /home/manu/Documents/PlatformIO/Projects/sample\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Traceback (most recent call last):\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/connectionpool.py\", line 677, in urlopen\n chunked=chunked,\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/connectionpool.py\", line 381, in _make_request\n self._validate_conn(conn)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/connectionpool.py\", line 976, in _validate_conn\n conn.connect()\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/connection.py\", line 370, in connect\n ssl_context=context,\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/util/ssl_.py\", line 377, in ssl_wrap_socket\n return context.wrap_socket(sock, server_hostname=server_hostname)\n File \"/usr/lib/python3.6/ssl.py\", line 407, in wrap_socket\n _context=self, _session=session)\n File \"/usr/lib/python3.6/ssl.py\", line 817, in __init__\n self.do_handshake()\n File \"/usr/lib/python3.6/ssl.py\", line 1077, in do_handshake\n self._sslobj.do_handshake()\n File \"/usr/lib/python3.6/ssl.py\", line 689, in do_handshake\n self._sslobj.do_handshake()\nssl.SSLError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/adapters.py\", line 449, in send\n timeout=timeout\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/connectionpool.py\", line 725, in urlopen\n method, url, error=e, _pool=self, _stacktrace=sys.exc_info()[2]\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/urllib3/util/retry.py\", line 439, in increment\n raise MaxRetryError(_pool, url, error or ResponseError(cause))\nurllib3.exceptions.MaxRetryError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /Wiz-IO/LIB/raw/master/quectel/toolchain-gcc-ec2x-v1.0.0.zip (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)'),))\n\nDuring handling of the above exception, another exception occurred:\n\nTraceback (most recent call last):\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/__main__.py\", line 109, in main\n cli() # pylint: disable=no-value-for-parameter\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 829, in __call__\n return self.main(*args, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/__init__.py\", line 44, in invoke\n return super(PlatformioCLI, self).invoke(ctx)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1259, in invoke\n return _process_result(sub_ctx.command.invoke(sub_ctx))\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/project.py\", line 143, in project_init\n pg.generate()\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/ide/projectgenerator.py\", line 140, in generate\n tpl_vars = self._load_tplvars()\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/ide/projectgenerator.py\", line 92, in _load_tplvars\n tpl_vars.update(load_project_ide_data(self.project_dir, self.env_name) or {})\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/project/helpers.py\", line 153, in load_project_ide_data\n raise result.exception\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/testing.py\", line 329, in invoke\n cli.main(args=args or (), prog_name=prog_name, **extra)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 782, in main\n rv = self.invoke(ctx)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 1066, in invoke\n return ctx.invoke(self.callback, **ctx.params)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/core.py\", line 610, in invoke\n return callback(*args, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/click/decorators.py\", line 21, in new_func\n return f(get_current_context(), *args, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/run/command.py\", line 138, in cli\n is_test_running,\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/run/command.py\", line 170, in process_env\n result = {\"env\": name, \"duration\": time(), \"succeeded\": ep.process()}\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/commands/run/processor.py\", line 79, in process\n result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/platform.py\", line 414, in run\n self.install_packages(silent=True)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/platform.py\", line 304, in install_packages\n \"%s=%s\" % (name, version), silent=silent, force=force\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/package.py\", line 702, in install\n pkg_dir = self._install_from_url(name, url, requirements, track=True)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/package.py\", line 480, in _install_from_url\n dlpath = self.download(url, tmp_dir, sha1)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/package.py\", line 195, in download\n raise e\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/managers/package.py\", line 179, in download\n fd = FileDownloader(url, dest_dir)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/platformio/downloader.py\", line 42, in __init__\n verify=sys.version_info >= (2, 7, 9),\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/api.py\", line 76, in get\n return request('get', url, params=params, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/api.py\", line 61, in request\n return session.request(method=method, url=url, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/sessions.py\", line 530, in request\n resp = self.send(prep, **send_kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/sessions.py\", line 643, in send\n r = adapter.send(request, **kwargs)\n File \"/home/manu/.platformio/penv/lib/python3.6/site-packages/requests/adapters.py\", line 514, in send\n raise SSLError(e, request=request)\nrequests.exceptions.SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /Wiz-IO/LIB/raw/master/quectel/toolchain-gcc-ec2x-v1.0.0.zip (Caused by SSLError(SSLError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:852)'),))\n\n============================================================\n\nAn unexpected error occurred. Further steps:\n\n* Verify that you have the latest version of PlatformIO using\n `pip install -U platformio` command\n\n* Try to find answer in FAQ Troubleshooting section\n https://docs.platformio.org/page/faq.html\n\n* Report this problem to the developers\n https://github.com/platformio/platformio-core/issues\n\n===================

Wiz-IO, 2020-07-03:
what is the question?

manusoman92, 2020-07-03:
I am getting this issue , while creating project on EC25, I am new to platformIO and VScode

Wiz-IO, 2020-07-03:
https://github.com/Wiz-IO/platform-quectel/wiki/Framework-OpenLinux#install-notes

manusoman92, 2020-07-03:
package.json under path platformio/packages/framework-quectel contains following lines only 

{
    "name": "framework-quectel",
    "description": "Quectel Frameworks for PlatformIO",
    "url": "https://github.com/Wiz-IO/framework-quectel",
    "system": "*",
    "version": "2.1.01"
}

Wiz-IO, 2020-07-03:
install or use cross GCC compiler for Linux ( float-soft )
and you need ADB "uploader" under Linux
PIO is python - work onall platforms...

closed 2020-07-06 by Wiz-IO

mrf1093, 2021-11-01:
hello Mr.Angelov
I've got the same problem([error.txt](https://github.com/Wiz-IO/framework-quectel/files/7455784/error.txt)) when I tried to create a new project using MC60 and Arduino framework with quectel platform.
I check the link: https://github.com/Wiz-IO/platform-quectel/wiki/Framework-OpenLinux#install-notes and install gcc-linaro_VERSION_arm-linux-gnueabi and changed package.json file path but the problem remains.
thank you for your kindly response in advanced

.