Wiz-IO/framework-quectel #11: Strange crash/hang on first call to __init_array_start[i]()

ASL07, 2020-08-05:
Hi Georgi,

BC66 SDK15 
Firmware BC66NBR01A10

I am having a very strange issue, it doesn't happen if I use your `BC66_CPP_HELLO` example but it it happening in my application. I have checked the contents of the config directory in the project root and they are the same between both projects. I have not modified any of the files of this repo except for the one I show below.

Any idea what can be causing this?

`__libc_init_array()` (called by os_init()) crashes when it first calls `__init_array_start[i]()`

Here are the contents of my main.cpp:
```C
#include <os_wizio.h>

#ifdef __cplusplus
extern "C" 
{
#endif

    #include "ql_trace.h"
    #include "ql_system.h"
    #include "ql_gpio.h"
    #include "ql_power.h"
    #include "ql_stdlib.h"
    #include "ql_error.h"
    #include "ql_uart.h"

    #include "custom_feature_def.h"
    #include "ql_common.h"
    #include "ql_type.h"
    #include "ql_timer.h"
    #include "ril_network.h"
    #include "ril_socket.h"
    #include "ril.h"
#ifdef __cplusplus
}
#endif

#include "app_uart.h"
#include "app_utils.h"
#include "app_debug.h"
#include "app_utils.h"
#include "app_state.h"
#include "app_events.h"
#include "sm.h"
#include "logger.h"

void proc_main_task(s32 task_id)
{
    ST_MSG msg;
    s32 ret;
 
// Using a LED for debugging  
Ql_GPIO_Init(PINNAME_NETLIGHT, PINDIRECTION_OUT, PINLEVEL_HIGH, PINPULLSEL_PULLUP);
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_LOW); 

    // Init OS (by Wiz-io)
    os_init();
}

```

I have also added code that toggles a GPIO LED in os_syscall.c, in order to see where it crashes, here is what i added:
```C
void __libc_init_array(void)
{
    size_t count;
    size_t i;
    count = __preinit_array_end - __preinit_array_start;
    for (i = 0; i < count; i++)
        __preinit_array_start[i]();
    _init();
    count = __init_array_end - __init_array_start;
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_HIGH);
Ql_Delay_ms(500);
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_LOW);
Ql_Delay_ms(500);

// It crashes or hangs here

    for (i = 0; i < count; i++)
    {
        __init_array_start[i]();

Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_HIGH);
Ql_Delay_ms(500);
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_LOW);
Ql_Delay_ms(500);
    }
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_HIGH);
Ql_Delay_ms(500);
Ql_GPIO_SetLevel(PINNAME_NETLIGHT, PINLEVEL_LOW);
Ql_Delay_ms(500);
}
```

Wiz-IO, 2020-08-05:
MAIN is C or CPP?

ASL07, 2020-08-05:
CPP

Wiz-IO, 2020-08-05:
[image]

ASL07, 2020-08-05:
Hi, 
Your example works for me, but my code doesn't, even though I haven't modified any files on the framework.
I will investigate, can you point me on to some direction towards what can cause `__init_array_start[i]();` to crash??

Wiz-IO, 2020-08-05:
__init_array_start() execute CPP class constructors

Wiz-IO, 2020-08-05:
with os_syscall.c, the malloc/free is "remaped" to Ql_MEM_Alloc/Ql_MEM_Free
so... maybe your project need this:
https://github.com/Wiz-IO/framework-quectel/blob/master/arduino/arduino/new.cpp

closed 2021-03-30 by Wiz-IO