Wiz-IO/framework-quectel #14: Usage with Espressif platform for ESP32 using Arduino Framework

alexqzd, 2020-10-19:
I'm having a hard time figuring out how can I use this as a library to connect with an EC25 over UART using an ESP32 with Arduino framework. Some guidance would be really appreciated. Thanks.

Wiz-IO, 2020-10-19:
Hi,

EC2x is a Linux module... this platform can compile normal linux/posix application ( executed inside module )
Ack your local Quectel support for all documents....
example: https://www.youtube.com/watch?v=sbLJNXTRLxc&ab_channel=GeorgiAngelov
_BTW: Arduino port is not shared yet..._

for ESP as master you can use any RIL ( AT commands ) Arduino library or
implement any open source RIL as
https://github.com/aosp-mirror/platform_hardware_ril/tree/master/reference-ril

Wiz-IO, 2020-10-19:
and read this
https://github.com/Wiz-IO/platform-quectel/wiki/Framework-OpenLinux

alexqzd, 2020-10-19:
Got it. Thanks!

closed 2020-10-19 by alexqzd